ha7ilm/openwebrx #102: Change Center frequency, from webpage

lu9cbl, 2017-12-01:
I'm cant change the center frequency on the web page, if i want to, i need to change the config file in the console, and then restart the process.

Thanks!

ha7ilm, 2017-12-01:
That's on my TODO list, but I'm making progress slowly.

Cheers,

Andras, HA7ILM

lu9cbl, 2017-12-01:
If i can help you, let me know. Thanks

Giorgiofox, 2018-02-24:
Many peoples waiting this ;) I think this is a very important feature, please.

Thanks in advance
Giorgio

Gymnae, 2018-03-15:
I'd also would immediately deploy this for my HF+ instead of using SpyServer if frequency and some of the settings KiwiSDR offers are available in this package.

It's a real pitty that @jks-prv did not commit to this project after the fork.

Giorgiofox, 2018-05-18:
Any update for this?

ha7ilm, 2018-05-19:
No update yet.

Gymnae, 2018-09-26:
I see that there are some forks and wrappers trying to add this feature, but non really work. 
Has anyone found a way to allow frequency change in the frontend? This feature would be really great.

My AirSpyHF+ is so Windows centric, that using it via Network on MacOS or Linux is no fun - openwebrx is the better way to make use of this hardware.

npaisnel, 2018-12-15:
Yes, just to add my voice to this request
The ability to change centre frequency from the webgui without having to ssh in and edit the config file would make this perfect for me.

I just want to monitor three separate frequencies  -all AM.  119.45mhz, 125.200 and 134.675
I keep my sample rate down to 250khz, running on a Pi 

To have these frequencies in a drop down bookmark list would be great too

KC9S, 2019-06-22:
I agree.  This is excellent software.  Really looking forward to center frequency tuning.  That’s really the only feature that I desire.  Let me know if I can help!

tarator, 2019-12-15:
+1 for this feature

Gymnae, 2019-12-16:
I've added it to my fork, based on the work of yet another fork :) It works nicely with Airspy and sdrplay devices in addition to the classis rtl-sdr:
https://github.com/gymnae/openwebrx