ha7ilm/openwebrx #105: Play Center Frequency To Local Speakers

roblevesque, 2018-01-03:
I have a use case where I'd like to use openwebrx but also play the center frequency to the hosting machine's local speakers. Is there a way to do this currently, or is it a easy feature to add?

Thanks!

dj1an, 2018-01-04:
On the machine running openwebrx you could do something like
``` bash
nc -v 127.0.0.1 4951 |
csdr through |
csdr convert_u8_f |
csdr shift_addition_cc $(python -c "print float(145100000-144800000)/2400000") |
csdr fir_decimate_cc 50 0.005 HAMMING |
csdr fmdemod_quadri_cf |
csdr limit_ff |
csdr deemphasis_nfm_ff 48000 |
csdr fastagc_ff |
csdr convert_f_s16 | 
mplayer -cache 1024 -quiet -rawaudio samplesize=2:channels=1:rate=48000 -demuxer rawaudio -
```
More examples on:
https://github.com/simonyiszk/csdr#usage-by-example

roblevesque, 2018-01-04:
Awesome. Thank you! 

---- On Thu, 04 Jan 2018 10:41:06 -0500 dj1an &lt;notifications@github.com&gt; wrote ----

On the machine running openwebrx you could do something like

nc -v 127.0.0.1 4951 | csdr through | csdr convert_u8_f | csdr shift_addition_cc $(python -c "print float(145100000-144800000)/2400000") | csdr fir_decimate_cc 50 0.005 HAMMING | csdr fmdemod_quadri_cf | csdr limit_ff | csdr deemphasis_nfm_ff 48000 | csdr fastagc_ff | csdr convert_f_s16 | mplayer -cache 1024 -quiet -rawaudio samplesize=2:channels=1:rate=48000 -demuxer rawaudio -

More examples on:

 https://github.com/simonyiszk/csdr#usage-by-example

closed 2019-12-31 by roblevesque