ha7ilm/openwebrx #106: Service for waterfall crashes all the time

pe1chl, 2018-01-05:
I installed OpenWebRx on a PC with SDRPlay RSP1 and I find that it works the first time
I try listening, but when the connection to the user is not so good the service for the waterfall
crashes with the messages shown below and when the next user connects there is no waterfall
anymore, but the receiver still sends audio and is tunable.

Exception happened during processing of request from ('(ip address here)', 34756)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 599, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe

pe1chl, 2018-01-13:
Any idea what this can be?  I see other sites running openwebrx apparently successfully but for us it cannot run more than an hour or so.
The system is Debian Jessie with its standard python 2.7 is that not OK?

ha7ilm, 2018-01-13:
Is this the only error? When a client disconnects, currently it is OK if it shows an exception like that.

pe1chl, 2018-01-13:
Yes this is the error we see on the console, but when the next client connects he gets only audio and no waterfall until the program is stopped and restarted.

pe1chl, 2018-01-13:
After another test with a client that just closes the browser I see this:

[openwebrx-httpd:ws] exception:  <class 'socket.error'> [Errno 32] Broken pipe
  File "./openwebrx.py", line 488, in do_GET
    rxws.send(self, temp_audio_data, "AUD ")
  File "/home/scoop/Downloads/openwebrx/rxws.py", line 164, in send
    myself.wfile.write(header+data_to_send)
  File "/usr/lib/python2.7/socket.py", line 324, in write
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
[openwebrx-httpd] client 44.137.41.97#8da419eaa1862bbe588ba171b1774402 :: client being closed.
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_140278170259536_bpf_pipe'
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_140278170259536_shift_pipe'
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_140278170259536_squelch_pipe'
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_140278170259536_smeter_pipe'
----------------------------------------
Exception happened during processing of request from ('44.137.41.97', 45157)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 599, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 657, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 716, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 279, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
----------------------------------------

However this time the program still works OK after reconnecting.  I have rebooted the system earlier today so maybe the problem has solved itself by magic...

ha7ilm, 2018-01-13:
OK, then please send the full console output of OpenWebRX if you encounter this again.