ha7ilm/openwebrx #109: high cpu usage with no client connected Ubuntu server 16.04

iw5ejm, 2018-02-08:
Is normal that the csdr process is running also with no client connected and using cpu as if there was clients?
[image]

ha7ilm, 2018-02-08:
Could you please tell the full command with `top -c`? If it is `csdr fft_cc, then it is okay.

iw5ejm, 2018-02-09:
[image]

ha7ilm, 2018-02-09:
This is normal. The `csdr fft_cc` process runs always, even if there is no user connected. This is which generates the waterfall diagram. It was easier to implement this way (although starting/stopping this process could be done).

closed 2018-02-09 by ha7ilm

iw5ejm, 2018-02-09:
Ok so to try stopping the fft_cc process I have to look in csdr.py file, DSP class right?

ha7ilm, 2018-02-09:
Yes.

ha7ilm, 2018-02-09:
You should look for spectrum thread.