ha7ilm/openwebrx #111: NOAA APT bandwidth

AlanJan, 2018-03-04:
Hi,
I have made a unsuccesfull attempt to modify the NFM demodulator to make it work for NOAA satellites @137 Mhz. I need to change the bandwidth to 36Khz to receive the APT signal and route it to WXtoImg.
Do have someone have any tips, to make it work?

cheers!

denadaisicari, 2018-03-11:
Im behind this issue too! Whats your location? Im trying one @ cerquilhosdr.com.br

AlanJan, 2018-03-12:
My location is in the Netherlands but I have take OpenWebRX offline. In meanwhile I have been switch to ShinySDR, (also A webbased radio for RTL-SDR )there is more support.Now I have been able to receive the APT signals for Polar orbiting satellites. You can simple add a Modulator for APT Signals.
but unfortunately here is no support for the Raspberry Pi, its runs on fine on Intel platforms like quadcore with 1Gb of ram.