ha7ilm/openwebrx #115: Audio receiving hangs

naorunaoru, 2018-06-05:
So I'm now running a test setup here: http://naoruatho.me:8073/

Currently it's tuned up on the Belorussky railway station frequency in Moscow. 

The server is Cubieboard2 with Cubian installed. Performance isn't very good, I'm getting low enough CPU usage with only 300ksps bandwidth. 

Everything runs ok, but from time to time it starts repeating a sample over and over and only reloading the page helps. Buffer overrun starts flashing. Waterfall still works. Other clients usually work fine when it happens, only to get the stutter some time later. 

I've turned off ADPCM audio compression hoping that it's a simple codec issue but no dice. 

Clients use latest Chrome on Windows PCs and Macs. On Edge sound just stops instead of looping. 

Can you help me debug this issue?

naorunaoru, 2018-06-05:
I've compiled `csdr` with following flags:

`PARAMS_NEON = -mfloat-abi=hard -mcpu=cortex-a7 -mtune=cortex-a7 -mfpu=neon-vfpv4 -mvectorize-with-neon-quad -funsafe-math-optimizations -Wformat=0 -ffast-math -DNEON_OPTS`

ha7ilm, 2018-06-06:
Could you please check the Javascript console in Chrome when it happens?

ha7ilm, 2018-06-06:
(Actually, the page can hang due to high memory usage if you leave it open for a very long time, e.g. hours or days. In the current version the waterfall is never cleaned up in the browser.)

naorunaoru, 2018-06-06:
@ha7ilm console output:

```
openwebrx.js:1475 audio overrun, 1
openwebrx.js:1475 audio overrun, 2
openwebrx.js:1475 audio overrun, 3
openwebrx.js:1475 audio overrun, 4
openwebrx.js:1475 audio overrun, 5
openwebrx.js:1302 waterfall queue length: 25
openwebrx.js:1475 audio overrun, 6
openwebrx.js:1302 waterfall queue length: 24
openwebrx.js:1475 audio overrun, 7
openwebrx.js:1475 audio overrun, 8
openwebrx.js:1302 waterfall queue length: 25
openwebrx.js:1475 audio overrun, 9
openwebrx.js:1302 waterfall queue length: 24
openwebrx.js:1475 audio overrun, 10
openwebrx.js:1475 audio overrun, 11
openwebrx.js:1302 waterfall queue length: 25
openwebrx.js:1475 audio overrun, 12
openwebrx.js:1302 waterfall queue length: 24
openwebrx.js:1475 audio overrun, 13
openwebrx.js:1475 audio overrun, 14
openwebrx.js:1302 waterfall queue length: 22
openwebrx.js:1475 audio overrun, 15
openwebrx.js:1475 audio overrun, 16
openwebrx.js:1475 audio overrun, 17
openwebrx.js:1475 audio overrun, 18
openwebrx.js:1475 audio overrun, 19
openwebrx.js:1475 audio overrun, 20
```

I found that the probability of getting stuck audio increases with server CPU load and does not correlate with how long the page has been open.

naorunaoru, 2018-06-06:
I fully understand that you shouldn't expect perfect audio with this setup, but at least the application should recover and not hang with a need to manually restart it.

ha7ilm, 2018-06-06:
Thanks for the feedback. Unfortunately I don't know yet why this happens at you. The application is designed to recover. You could probabily check the internal state of the application by sending it the USR1 or USR2 signal (I don't recall which). There's a particular variable called `loopstat` that shows where the server stays if there is a hang.

naorunaoru, 2018-06-06:
I'm guessing that it has something to do with client audio processing, not the server. I'll try to look into it some time later.