ha7ilm/openwebrx #117: no waterfall, audio and broken pipe

KacperStachula, 2018-09-06:
Hi, 
I want to share the receiver but I have problems.

I use a gray rtl sdr attached to a computer with Ubuntu
./openwebrx.py
`OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-main] nmux_bufsize = 12288, nmux_bufcnt = 4070
[openwebrx-main] Started rtl_thread: rtl_sdr -s 48000 -f 52000000 -p 0 -g 5 -| nmux --bufsize 12288 --bufcnt 4070 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0x99edf0: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2400 | csdr logaveragepower_cf -70 4096 2 | csdr fft_exchange_sides_ff 4096
[openwebrx-main] Starting HTTP server.
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
[openwebrx-spectrum] Spectrum thread started.

Using device 0: Generic RTL2832U OEM
nmux: pthread_create() done, clients now: 2
client 0xb13a80: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!

Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
Failed to open rtlsdr device #0.
nmux: (main thread/for) end input stream, exiting.
`
Now I'm going to my local site http://192.168.0.101:8073/

I do not see the spectrum I can not hear the signal and I see it in the log

`OpenWebRX client log
Author: András Retzler, HA7ILM
You can support OpenWebRX development via PayPal!
WebSocket opened to ws://192.168.0.101:8073/ws/
Server acknowledged WebSocket connection.
Audio stream is uncompressed.
FFT stream is uncompressed.
Web Audio API succesfully initialized, sample rate: 48000 sps
WebSocket has closed unexpectedly. Please reload the page.`

while in the console in Linux I see
`OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-main] nmux_bufsize = 12288, nmux_bufcnt = 4070
[openwebrx-main] Started rtl_thread: rtl_sdr -s 48000 -f 52000000 -p 0 -g 5 -| nmux --bufsize 12288 --bufcnt 4070 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0x99edf0: started!
[... 70 lines cut ...]
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr shift_addition_cc --fifo /tmp/openwebrx_pipe_3078624556_shift_pipe | csdr fir_decimate_cc 3 0.05 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_3078624556_bpf_pipe 0.02 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_3078624556_squelch_pipe --outfifo /tmp/openwebrx_pipe_3078624556_smeter_pipe 5 1 | csdr fmdemod_quadri_cf | csdr limit_ff | csdr old_fractional_decimator_ff 1.33333333333 | csdr deemphasis_nfm_ff 11025 | csdr fastagc_ff 1024 | csdr convert_f_s16
nc: connect to 127.0.0.1 port 4951 (tcp) failed: Connection refused
csdr shift_addition_cc: fir_decimate_cc: taps_length = 79
fifo control mode on
csdr old_fractional_decimator_ff: window = HAMMING
csdr old_fractional_decimator_ff: taps_length = 133
csdr squelch_and_smeter_cc: fifo control mode on
csdr bandpass_fir_fft_cc: fifo control mode on
csdr shift_addition_cc: reinitialized to -0
csdr squelch_and_smeter_cc: initial squelch level is 0
[openwebrx-httpd:ws,0] command: SET mod=nfm low_cut=-4000 high_cut=4000 offset_freq=0
[openwebrx-httpd:ws] exception:  <type 'exceptions.IOError'> [Errno 32] Broken pipe
  File "./openwebrx.py", line 563, in do_GET
    dsp.set_offset_freq(int(param_value))
  File "/home/init/openwebrx/csdr.py", line 268, in set_offset_freq
    self.shift_pipe_file.flush()
[openwebrx-httpd] client 192.168.0.100#ac20a6b5fecdf78030aeb6f0fac78315 :: client being closed.
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_3078624556_bpf_pipe'
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_3078624556_shift_pipe'
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_3078624556_squelch_pipe'
[openwebrx-dsp-plugin:csdr] try_delete_pipes() :: [Errno 2] No such file or directory: '/tmp/openwebrx_pipe_3078624556_smeter_pipe'
  
`
thanks for the answers
SP1ZZW

ha7ilm, 2018-09-07:
Thank you for your feedback. Which exact browser version are you using?

Andras

KacperStachula, 2018-09-07:
I'm using Chrome version 69.0.3497.81 64 but the problem also appears in firefox
As soon as we fix the problem, we provide the receiver on the network

Kacper 
SP1ZZW

ha7ilm, 2018-09-07:
Looks like this is the problem:

```
Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

usb_claim_interface error -6
Failed to open rtlsdr device #0.
```

Please check the setup guide especially about blacklisting the `dvb_usb_rtl28xxu` module: https://blog.sdr.hu/2015/06/30/quick-setup-openwebrx.html

KacperStachula, 2018-09-07:
thanks, everything works as soon as this month sharing in sdr.hu

Kacper 
SP1ZZW

closed 2018-09-07 by KacperStachula