ha7ilm/openwebrx #122: Openwebrx and direct sampling mode crash

sm6hoc, 2019-01-14:
Hi my friends
I'w got a problem wen install direct sampling mode. When install openwebrx everything is ok up and running.
The problem is when I install  the direct sampling mode (Have install 3 other and no problem). This is the
first Ubuntu 18.04. I copy the log here but it is only from startup script.

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-main] nmux_bufsize = 602112, nmux_bufcnt = 84
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2400000 -f 144250000 -p 0 -g 5 -| nmux --bufsize 602112 --bufcnt
84 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
rtl_sdr: symbol lookup error: rtl_sdr: undefined symbol: rtlsdr_set_dithering
nmux: listening on 127.0.0.1:4951
nmux: (main thread/for) end input stream, exiting.

Lars SM6HOC

ha7ilm, 2019-01-15:
> rtl_sdr: symbol lookup error: rtl_sdr: undefined symbol: rtlsdr_set_dithering

Yes, probably there is a mismatch between your `librtlsdr` library and `rtl_sdr` tool version. Try to remove all RTL-SDR installations (both the one installed through the package manager and the one installed via command line), and reinstall keenerd's fork.

sm6hoc, 2019-01-15:
Hi again !
Yes I doing this BUT not reinstall "Keennerds Fork. Now I have done this (from begining to the end...) and now its running. The server is a IBM x346.
Thank you
Lars SM6HOC
----Ursprungligt meddelande----

ha7ilm, 2019-01-19:
All right! 

AFAIK then the problem is solved by now.

closed 2019-01-19 by ha7ilm