ha7ilm/openwebrx #126: Attach audio element to document tree, please

JB-DX, 2019-02-16:
I need to route the web browser audio to a specific device, not the default device. This can be done by using the chrome extension Audio-Pick https://rain-fighters.github.io/AudioPick/ 
Unfortunately this does not work  with openwebrx. Probably because the  audio element is not within the document tree as the AudioPick reademe  suggest.
Q:
Could the audio element be attached to the document tree please?

dj1an, 2019-02-16:
On Linux you could use pulseaudio or alsa loop for this.
On Windows there are also some "Virtual audio cable" solutions available, to route the sound from one Device to another.

JB-DX, 2019-02-16:
On Windows I intend to route it via VAC to another application (Spectrum Lab). For that you need to be able to select the input of VAC. But the browser (or openwebrx) does not provide that feature. Only Audio-Pick allows it, but not for openwebrx.
Note: I just want to route  Kiwi  to VAC, not the entire browser or even default windows audio.

ha7ilm, 2019-02-17:
Unfortunately there is no `<audio>` element at all. OpenWebRX uses the Web Audio API, so AudioPick wouldn't work with it anyway. 

Anyway, the KiwiSDR software is different from OpenWebRX (although it is based on that), so what I say does not apply for sure.

I think on Windows you could try to set the default audio device to VAC, at least while you start Chrome.

ha7ilm, 2019-03-03:
As suggested above, you should try to install Virtual Audio Cable, set the default audio device to Virtual Audio Cable here:

[image]

...and then restart Google Chrome, and open a receiver.

[Source of the image](https://www.raymond.cc/blog/virtual-audio-cable-enables-multiple-audio-outputs-in-windows-7/)

I'm closing this now.

closed 2019-03-03 by ha7ilm