ha7ilm/openwebrx #127: websocket hang?

canthony15, 2019-02-20:
I am creating a modified version of openwebrx tailored more to my NDB hobby. I have it running on the local network but I ran into an issue when I set up port forwarding to the outside. Running from the outside, everything is OK for about 8 minutes and then waterfall and audio stop. After a while the server says that the spectrum queue is full and closes the connection. My connection is low performance at my home location with less than 1Mb upload but network usage shows less than 300K. To be sure, I disabled audio by not sending the start message and set the fft_fps to 1. It still fails at about 8 minutes. It appears to be the connection time rather than a queue filling up somewhere. The Chrome console shows the WS stream just stopping.

Anyway, I am thinking that this might be related to my ISP (Centurylink) but I thought it would be worth bringing up here so that others who might see this sort of issue will know that they are not alone.  If you have seen anything like this I would be interested in any information you might have. I have gone through the code and I cannot see anything that would make this happen. I am inexperienced with websockets and such so I could be missing something.

canthony15, 2019-02-20:
I should also mention that the radio was eliminated by replacing it with a random number generator and pv in the config file. The radio itself is totally out of the picture. The problem did not change.

ha7ilm, 2019-02-20:
Check if you run out of memory with Chrome (although that is not likely to happen after 8 minutes).

canthony15, 2019-02-20:
I did that and the memory is fine. To be sure I commented out the push to the waterfall queue and it still fails.

ha7ilm, 2019-02-20:
Thank you for your report. I don't know then why it fails.

canthony15 <notifications@github.com> ezt írta (időpont: 2019. febr. 20.,
Sze 18:42):

canthony15, 2019-02-20:
Here is an interesting capture from the chrome console.  The stream stalls at 120558. The server then closes the connection because the spectrum queue fills. But then, at 120921, a spew of frames that must have been queued somewhere spills out. For someone more saavy than me, this probably means something.
[image]

canthony15, 2019-02-21:
Update. It does not fail when running on chrome in my old samsung S3 phone.

canthony15, 2019-02-23:
The problem was solved by replacing the DSL modem (Actiontec Q1000) with a Zyxel unit.  I will close this now.

closed 2019-02-23 by canthony15