ha7ilm/openwebrx #137: Modification for python3

oe2lsp, 2019-09-15:
according to https://pythonclock.org/ python2[.7] is going to be retired at the end of the year. Is there any way we can get an python3 compatible version of the great openwebrx?

thanks
Lucas

ha7ilm, 2019-09-15:
Yes, I plan to finish porting OpenWebRX to Python 3 by that time.

oe2lsp, 2019-09-15:
great to hear!

jketterl, 2019-10-03:
I have taken an afternoon and ported the original code over to python 3 here: https://github.com/jketterl/openwebrx/tree/python3

I made it to the point where it seems to be mostly working, but it certainly can use some more testing.

Also, this is not downwards compatible. I doubt that it is possible without plenty of overhead... but since python 2 should be close to it's end, I don't think it's much of a problem.

D0han, 2019-10-03:
I have been running @jketterl version under py3 for months already, can recommend.

jketterl, 2019-10-03:
@D0han that's not the version I linked above. This is a python3 port of the original code.

ha7ilm, 2019-10-04:
> I have taken an afternoon and ported the original code over to python 3 here: https://github.com/jketterl/openwebrx/tree/python3

Thanks, I'll take a look.

ha7ilm, 2019-12-29:
> > I have taken an afternoon and ported the original code over to python 3 here: https://github.com/jketterl/openwebrx/tree/python3
> 
> Thanks, I'll take a look.

In the meantime, I've decided to discontinue the whole project, so I don't plan to support Python 3 myself. On this reason, I'm closing this.

closed 2019-12-29 by ha7ilm