ha7ilm/openwebrx #141: Error code explanation: 404 = Nothing matches the given URI.

Podly27, 2019-10-19:
Hello,

i installed openwebrx on 2 hw (RaspberryPI 3 and laptop) 2 os (Raspbian and LinuxMint) but OpenWebRX still does not work. Could you help me ?

commandline:
pi@raspberrypi:~ $ openwebrx/openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Andras Retzler, HA7ILM <randras@sdr.hu>

[openwebrx-main] Configuration script not specified. I will use: "config_webrx.py"
[openwebrx-main] nmux_bufsize = 602112, nmux_bufcnt = 84
[openwebrx-main] Started rtl_thread: rtl_sdr -s 2400000 -f 144250000 -p 0 -g 5 -| nmux --bufsize 602112 --bufcnt 84 --port 4951 --address 127.0.0.1
[openwebrx-main] Waiting for I/Q server to start...
nmux: listening on 127.0.0.1:4951
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
[openwebrx-main] I/Q server started.
[openwebrx-main] Starting watchdog threads.
nmux: pthread_create() done, clients now: 1
client 0x1249680: started!
[openwebrx-main] Starting spectrum thread.
[openwebrx-main] Starting HTTP server.
[openwebrx-spectrum] Spectrum thread initialized successfully.
[openwebrx-dsp-plugin:csdr] Command = nc -v 127.0.0.1 4951 | csdr convert_u8_f | csdr fft_cc 4096 2867 | csdr logaveragepower_cf -70 4096 93 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
[openwebrx-spectrum] Spectrum thread started.
nmux: pthread_create() done, clients now: 2
client 0x1249748: started!
Connection to 127.0.0.1 4951 port [tcp/*] succeeded!
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 144250000 Hz.
Tuner gain set to 3.70 dB.
Reading samples in async mode...
Allocating 15 zero-copy buffers
client 0x1249680: CS_THREAD_FINISHED, client_goto_source = 2, errno = 32192.168.1.102 - - [19/Oct/2019 08:58:21] code 404, message Invalid path.
192.168.1.102 - - [19/Oct/2019 08:58:21] "GET / HTTP/1.1" 404 -
192.168.1.102 - - [19/Oct/2019 08:59:01] code 404, message Invalid path.
192.168.1.102 - - [19/Oct/2019 08:59:01] "GET / HTTP/1.1" 404 -
192.168.1.102 - - [19/Oct/2019 08:59:02] code 404, message Invalid path.
192.168.1.102 - - [19/Oct/2019 08:59:02] "GET / HTTP/1.1" 404 -
nmux: signal 2 caught, exiting...
[openwebrx] Ctrl+C: aborting.
Signal caught, exiting!

Web browser:

Error response
Error code 404.
Message: Invalid path..
Error code explanation: 404 = Nothing matches the given URI. 

.../status
status=active
name=[Callsign]
sdr_hw=RTL-SDR
op_email=example@example.com
bands=143050000-145450000
users=0
users_max=20
avatar_ctime=
gps=(47.0, 19.0)
asl=200
loc=Budapest, Hungary
sw_version=v0.17
antenna=Longwire

Thanks for help.

ha7ilm, 2019-10-19:
Are you running OpenWebRX so that the CWD is its directory?

Podly27 <notifications@github.com> ezt írta (időpont: 2019. okt. 19., Szo
9:03):

Podly27, 2019-10-19:
Resolved :-)

NOT WORKING
pi@raspberrypi:~ $ openwebrx/openwebrx.py

WORKING
pi@raspberrypi:~ $ cd openwebrx
pi@raspberrypi:~ $ ./openwebrx.py

Grrrrrrr. Stupid mistake

closed 2019-10-19 by Podly27