ha7ilm/openwebrx #142: Accessibility, Please?

voipworld, 2019-12-15:
Hello Developer,
i am 7X2SW (Tayeb) from algeria
i have a RTL-SDR V3 PRO, and i prefer to use a web app unstead of a desktop app, i found OpenWebRX (topnotch sdr receiver)
but please, i dont see anywhere how to change the freq & bands, using a screen reader
may you please try to use html5 from elements to implement frequency change, in adition to the nice graphical user interface allready made?
radio is friend of blinds, btw ;)
thanks,
Tayeb.

dj1an, 2019-12-16:
Hello Tayeb,
it's not possible in openwebrx to change the Center Frequency/Band while running. 
There exists a Fork from @jketterl on https://github.com/jketterl/openwebrx which implements this Function.
A while ago i created a Pull Request #104  to tune Frequency by Mouse and Keyboard. Changing Modes by Key combinations should be easy to implement. 

73 de DJ1AN

voipworld, 2019-12-16:
Hi,
i think you missunderstand my issue?
or maybe i didnt explain it correctly?
for example, the OpenWebRX is tuning by default into 144.70mhz, what
about if i want to switch to 145.500MHZ?
i think, there exist a way in the ui to do it, no?
but, using my screen reader, i cant find any control for tuning
thanks,
73S from 7X2SW.

dj1an, 2019-12-16:
Hi,
there is no control to directly input a Frequency. 
Normally you just click into the waterfall and it will jump to there.
73 de DJ1AN, Jan

jketterl, 2019-12-16:
Hello,

i think this is the first time somebody has requested to change the frequency without implying the center_freq :grinning: Now, I think I understand the request, and given the primary graphical nature of the application, I also understand that it must be very hard to use with impaired vision.

I'm not sure if it's a good idea to try and implement the accessibility features into the current user interface, but I could very well imagine building a separate, text-based version, where all the inputs are properly labeled. This might have a waterfall display, but more for illustration purposes.

Also, I'm not sure how, but maybe there's an easy way to make bookmarks (new feature on my fork) available to screen readers. That way, you could at least tune into the stations that have been identified by the receiver admin.

What do you think?

73s Jakob DD5JFK

voipworld, 2019-12-17:
Hey,
maybe i am expressing the issue in a wrong way?
please may you take a look at http://www.websdr.org?
i can use it without issue at all, there's + and - buttons, to make up
& down the freq
get it?

dj1an, 2019-12-18:
Hello Tayeb,
websdr.org is not openwebrx.
It's a closed source project from pa3fwm.
So send your Feature Requests and issues directly to him.

Above the +- buttons is a frequency input field where you can directly enter a frequency, if this helps.
73 de DJ1AN

jketterl, 2019-12-18:
I think the point is not to get us to fix something in websdr... The point is to highlight that websdr is a good example for accessibility :wink:

voipworld, 2020-01-05:
hey,
i think you still didnt get my point at all
i am not complaining about websdr itself, but i sayd its a good software to be used by blind users. i gave websdr as a accessible sdr client example
i am asking, how to use the OpenWebRX similarly to webSDR?
because in OpenWebRX, there's no buttons, is all graphics.